bitbank2/OneBitDisplay #4: 1px rendering error using obdWriteString

cyberman54, 2020-05-16:
I am using `obdWriteString()` to display text on a SSD1306 oled display, which is refreshed with 25fps. I oberved a slightly line flickering, and discovered that this is caused by sometimes a line with text is shifted 1xp back and forth (to left, than back to right).

Since i'm using the same code with former ssoled library, i guess this behaviour is caused by the library.

bitbank2, 2020-05-18:
Please provide sample code to reproduce this problem. I have not seen this problem on the latest code.

cyberman54, 2020-05-18:
https://github.com/cyberman54/ESP32-Paxcounter/blob/development/src/display.cpp

bitbank2, 2020-05-18:
Please provide a SIMPLE/MINIMAL example which shows the problem. Your code above is several layers deep and is inconclusive.

closed 2020-05-18 by bitbank2