bitbank2/OneBitDisplay #5: Buffered text with obdWriteString doesn't work on the lower half of a 128x128 display

ed-wigg, 2020-05-18:
`obdWriteDataBlock` has 1024 hard coded as the maximum length of the screen buffer so buffered text the lower half of a 128x128 display is ignored. I fixed it in my case by increasing the maximum length to 2048 (and setting the bit mask to 2047), but there may be a better solution.

bitbank2, 2020-05-18:
I just made major changes to the library to support virtual displays of any size. This required that I fix all of the areas which had hard coded values of 1023. Please do a git pull from the experimental branch and verify that it's fixed. I will be doing an official release shortly of this latest code.

closed 2020-05-18 by bitbank2

ed-wigg, 2020-05-18:
The experimental branch works correctly.

bitbank2, 2020-05-18:
Great - thanks for letting me know.