bitbank2/OneBitDisplay #7: Refresh display value without obdFill?

Swiftnesses, 2020-05-29:
Hello,

Thanks for the lib - I've been trying to find something that works with my display (this is the first that works perfect). I'm looking to have a screen that displays various moisture levels for plants, in the format of:

Zone 1: 34%
Zone 2: 56%

I'd like to update the number in real-time. Currently I'm doing this with a delay and `obdFill(&obd, 0, 1)`, but I'm sure there's a better way (previous with u8g I just set the coordinates and changed the value, cursor?).

Any help of advice is greatly appreciated!

TalibaniGrannyFanny, 2020-05-29:
What display module are you using for your project...?

Swiftnesses, 2020-05-29:
Hi,

I believe it's a SH1106 based display...

TalibaniGrannyFanny, 2020-05-29:
do you mean the 128x64 OLED module, do you....?

Swiftnesses, 2020-05-29:
Hi GrannyFanny 🥇 

Yes, it's a 128x64 module :)

TalibaniGrannyFanny, 2020-05-29:
do you have the spi or i2c version of the board?

TalibaniGrannyFanny, 2020-05-29:
/*
give this sketch a go (it's actually for displaying the temp values on a whole RANGE of LCD displays and OLEDs {both i2C & SPI...} in the format of bargraph, they are updated real-time, as requested....

LET ME KNOW how you go with it, or ask if any of it isn't clear to you, and Ill do my veryvery best to help you along :)

*/

#include <Arduino.h>
#include <Wire.h>
#include <LCD.h>

#define _LCD_SR3W_

#ifdef _LCD_I2C_
#include <LiquidCrystal_I2C.h>
#endif

#ifdef _LCD_4BIT_
#include <LiquidCrystal.h>
#endif

#ifdef _LCD_SR_
#include <LiquidCrystal_SR.h>
#endif

#ifdef _LCD_SR3W_
#include <LiquidCrystal_SR3W.h>
#endif

/*!
    @defined    CHAR_WIDTH
    @abstract   Character witdth of the display, expressed in pixeles per character.
*/
#define CHAR_WIDTH  5

/*!
    @defined    BACKLIGHT_PIN
    @abstract   LCD backlight pin definition.
    @discussion AVR pin used for the backlight illumintation of the LCD.
*/
#define BACKLIGHT_PIN          12

/*!
    @defined    STATUS_PIN
    @abstract   Status LED indicator.
    @discussion Activity LED blinking indicating that the system is up.
*/
#define STATUS_PIN             13

/*!
    @defined    LOOP_DELAY
    @abstract   Main loop delay.
    @discussion Main loop delay executing temperature readings and LCD updates.
*/
#define LOOP_DELAY            300

/*!
    @defined    TEMP_CAL_OFFSET
    @abstract   Temperature calibration offset.
    @discussion This is the offset value that has to be modified to get a
                correct temperature reading from the internal temperature sensor
                of your AVR.
*/
#define TEMP_CAL_OFFSET 282

/*!
    @defined    FILTER_ALP
    @abstract   Low pass filter alpha value
    @discussion This value defines how much does the current reading, influences
                the over all value. The smaller, the less influence the current
[... 216 lines cut ...]
}

void loop ()
{
  int temp;
  static byte status = 1;
  
  status ^= 1;
  digitalWrite ( STATUS_PIN, status);
  
  temp = readTemperature();
  tempFilter = ( FILTER_ALP * temp) + (( 1.0 - FILTER_ALP ) * tempFilter);

  // Display the information to the LCD
  myLCD->setCursor ( 8, 0 );
  myLCD->print ("     ");
  myLCD->setCursor ( 8, 0 );
  myLCD->print ( tempFilter, 1 );
  myLCD->setCursor ( 12, 0 );
  myLCD->print ( "\x07" );
  myLCD->print ("C");
  drawBars ( tempFilter, 1, 14, '-', '+' );
  
  delay (LOOP_DELAY);
}

TalibaniGrannyFanny, 2020-05-29:
/*
 * This Arduino sketch is to use LM35 to measure temperature

 * prints the temperature as C, F or K on the LCD1602 or LCD2004 with I2C module (4 wires)
 
 * 

 *  * This code is "AS IS" without warranty or liability. Free to be used as long as you keep this note intact.* 
 * This code has been download from Robojax.com
    This program is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    This program is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <https://www.gnu.org/licenses/>.

*/

const int inPin =A0;//can change
const int VCC2 =2;
const int iteration = 1000; //can change (see video)
const float LM35_FACTOR =0.01;// do not change

// start of settings for LCD1602 with I2C
#include <LiquidCrystal_I2C.h>
// Set the LCD address to 0x27 for a 16 chars and 2 line display
LiquidCrystal_I2C lcd(0x26, 16, 2);
// end of settings for LCD1602 with I2C

// the setup routine runs once when you press reset:
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
  
  pinMode(VCC2,OUTPUT);
  digitalWrite(VCC2, HIGH);
  
  lcd.begin();  
  lcd.backlight();
  lcd.print("Real-Time LM35");
  lcd.setCursor(0,1);
  lcd.print("Temp: "); 
}

// the loop routine runs over and over again forever:
void loop() {

  lcdDisplay(getTemperature('C'),'C');
  delay(2000);
  
  lcdDisplay(getTemperature('F'),'F');
  delay(2000);
    
  lcdDisplay(getTemperature('K'),'K');  
  delay(2000);    

if(getTemperature('C') >87)
{
 // do somethig here (watch video)
}

//  printTemperature('C');
//  Serial.println();
//  printTemperature('F');
//  Serial.println();  
//  printTemperature('K'); 
//  Serial.println();     
//  Serial.println();    
//  Serial.print(" Temperature: ");
//  printDegree();    
//  Serial.print(getTemperature('C'));
//  Serial.println();
  delay(100);
}

/*
 * getTemperature()
 * @brief gets the average temperature
 * @param average temperature
 * @param "type" is character
 *     C = Celsius
 *     K = Kelvin
 *     F = Fahrenheit
 * @return returns one of the values above
 */
float getTemperature(char type)
{
    float value;
    float averageTemperature =0;
    int sensorValue = analogRead(inPin);
    float voltage = sensorValue * (5.0 / 1023.0); 
    float temperature = voltage / LM35_FACTOR;
    for(int i=0; i< iteration; i++)
    {
      averageTemperature += temperature;
    }

    averageTemperature /=iteration;
  if(type =='F')
   {
    value = averageTemperature *9/5 + 32;//convert to Fahrenheit 
   }else if(type =='K')
   {
    value = averageTemperature + 273.15;//convert to Kelvin
   }else{
    value = averageTemperature;// return Celsius
   }    
    return value ;
}//getTemperature()

/*
 * printTemperature()
 * @brief prints  temperature on serial monitor
 * @param charact type
 * @param "type" is character
 *     C = Celsius
 *     K = Keliven
 *     F = Fahrenheit
 * @return none

 */
void printTemperature(char type)
{
    float value;
    float temp = getTemperature(type);
   Serial.print(temp);
    printDegree();    
  if(type =='F')
   {
     Serial.print("F");
    }else if(type =='K')
   {
     Serial.print("K");
   }else{
     Serial.print("C");
   }    
 
}//printTemperature()

/*
 * @brief prints degree symbol on serial monitor
 * @param none
 * @return returns nothing

 */
void printDegree()
{
    Serial.print("\xC2"); 
    Serial.print("\xB0");  
}

/*
 * lcdDisplay(float value,char symbol)
 * displays value and title on LCD1602
 * How to use:
 * lcdDisplay(35.3,'C');

 */
void lcdDisplay(float value,char symbol)
{

   for(int i=7; i<16;i++)
   {
    lcd.setCursor(i,1);
    lcd.write(254);  
   }
   lcd.setCursor (7,1); //
   lcd.print(value);
   lcd.print((char)223);    
   if(symbol =='F')
   {
    lcd.print("F");
   }else if(symbol =='K')
   {
    lcd.print("K");
   }else{
    lcd.print("C");    
   }
}

Swiftnesses, 2020-05-29:
Neither use your lib, correct?

bitbank2, 2020-06-21:
Sorry, I just saw this issue. You shouldn't need to fill the display to change the data. Just overwrite the same text and it will display correctly while not producing a "flash"

closed 2020-06-21 by bitbank2