bitbank2/OneBitDisplay #25: 128x32 random pixels of "snow"

gnbl, 2021-11-22:
Hi, 

I'm using PlatformIO with an ATtiny85@16.5 MHz (digispark clone) and Arduino / ATTinyCore (for SPI.h) with a generic 128x32 OLED module, that works with U8X8 (tinyd core).
Using the following code, the display is correctly detected, but I only get random pixels on the display instead of "Hello". 

;   bitbank2/BitBang_I2C @ ^2.1.5
;   bitbank2/OneBitDisplay @ ^1.11.0

    #include <BitBang_I2C.h>
    #include <OneBitDisplay.h>
    OBDISP obd;

    void setup()
    {
     
      int r = obdI2CInit(&obd, 
        OLED_128x32, 0,//0x3C, 
        0, 0, 
        0, 
        0xB2, // PB2
        0xB0, // PB0
        -1, 
        400000L);
    
      if(r<0) 
        digitalWrite(LED, HIGH);
      else{
        obdWriteString(&obd, 0,0,0, (char *)"Hello", FONT_NORMAL, 0, 1);
        digitalWrite(LED, HIGH);
        delay(100);
        digitalWrite(LED, LOW);
      }
    
      while(1);
    }

Any guesses where things could be going wrong?

gnbl, 2021-11-22:
Actually, it seems it does not detect the display.

gnbl, 2021-11-22:
I2CTest() looks like it returns 0 for __AVR_ATtiny85__ without even trying ;-)
It's also run twice in obdI2CInit() for some reason.
Afterwards, whatever is being sent, it's not the initialization data :-(
I'd try the tinyd core, but OBD requires <SPI.h> also for the bit-bang I2C display..

bitbank2, 2022-05-09:
The init function doesn't clear the display memory. Use obdFill() to clear it after initialization.

closed 2022-05-09 by bitbank2