bitbank2/OneBitDisplay #29: Something broke between v 1.11.0 and current (pre?) 2.0.0

cyberman54, 2022-04-25:
Board: Olimex ESP32 POE ISO with SSD1306 connected via I2C

```
- framework-arduinoespressif32 3.20001.0 (2.0.1)
<OneBitDisplay> 2.0.0+sha.0b84b4a        
|   |-- <BitBang_I2C> 2.1.5
|   |   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
```

=> Display SSD1306 is working, but showing only two lines on the top, and with wrong positioning of text in these lines

```
- framework-arduinoespressif32 3.20001.0 (2.0.1)
|-- <OneBitDisplay> 1.11.0
|   |-- <BitBang_I2C> 2.1.5
|   |   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
```

=> (same) Display SSD1306 is working properly (while it is not identified by I2CDiscoverDevice)

cyberman54, 2022-04-25:
added: I can reproduce this with a different board type (a "TTGO T-fox"), which has SSD1306 too, but from different source and with different look.

cyberman54, 2022-04-25:
See also https://github.com/cyberman54/ESP32-Paxcounter/issues/858

cyberman54, 2022-04-26:
Meanwhile i found a workaround for this issue: setting   `bbi2c.bWire = 1` (instead 0) before calling  `I2CInit(&bbi2c, 100000L)`. This makes the display to be identified with `I2CInit()` and after this Onebitdisplay is working properly.

cyberman54, 2022-05-02:
@bitbank2 any comments? 2.0.0 seems broken.

cyberman54, 2022-05-03:
Meanwhile i figured out, positioning got broken with [this commit](https://github.com/bitbank2/OneBitDisplay/commit/63d89ce4d7ca63523ffdeaf8d75090fa117990c7), and was not fixed by one of the later commits.

bitbank2, 2022-05-03:
I'm sorry, I need to document all of the changes when I added the C++ class. The x/y coordinates needed to behave more like TFT_eSPI/Adafruit_GFX, so yes the Y is now in pixels not lines.

closed 2022-05-03 by bitbank2

cyberman54, 2022-05-03:
What? This is major breaking change, that should have been documented somewhere. I thought this is library, not application software. It's hard to use it if suddenly breaking changes are to be expected.

But thanks for the note, it should be possible now to adapt my application.

bitbank2, 2022-05-03:
I apologize for the disturbance; you're welcome to continue using the older version of the library until I have updated the documentation.