bitbank2/OneBitDisplay #30: [DOUBT] How to clear the lines or display without blink

Girum, 2022-05-05:
Hi! When I use the library, everytime I want to clear the screen, a have to use the odbFill... but it blinks the enteire screen, even in 800kHz. How to print the string without use the fill ou how to avoid the blink screen?

bitbank2, 2022-05-09:
When not using a backbuffer, text will overwrite only the area written. If you want to blank a line, just write a line of spaces at that spot.

closed 2022-05-09 by bitbank2

joex92, 2022-05-14:
what if I need to display like Plotter Values, so each loop updates the values, I would need to clear the display every loop...

bitbank2, 2022-05-15:
The OLED can be overwritten without clearing it first. That will remove the flashing. You can do this without a back buffer, but it's easier to manage changing a few pixels at a time when you have a 1K buffer to hold the data.