bitbank2/OneBitDisplay #31: error compiling

80stech, 2022-05-06:
I am trying out OneBitDisplay (installed from board manager) but get these error messages. I've tried compiling for simple, simpleavr, multi demos and for Nano and Every but always get these error messages.  

Arduino:C:\\Users\\Rainer\\AppData\\Local\\Temp\\arduino_build_641990\\libraries\\OneBitDisplay\\OneBitDisplay.cpp.o"

In file included from C:\Users\Rainer\Documents\Arduino\libraries\OneBitDisplay\src\OneBitDisplay.cpp:46:0:

C:\Users\Rainer\Documents\Arduino\libraries\OneBitDisplay\src\OneBitDisplay.h: In constructor 'ONE_BIT_DISPLAY::ONE_BIT_DISPLAY()':

C:\Users\Rainer\Documents\Arduino\libraries\OneBitDisplay\src\OneBitDisplay.h:120:127: warning: overflow in implicit constant conversion [-Woverflow]

     ONE_BIT_DISPLAY() { memset(&_obd, 0, sizeof(_obd)); _obd.iFG = 1; _obd.render = 1; _obd.type = OLED_128x64; _obd.iSpeed = 400000;}

                                                                                                                               ^~~~~~

C:\Users\Rainer\Documents\Arduino\libraries\OneBitDisplay\src\OneBitDisplay.cpp: In function 'GFXglyph* pgm_read_glyph_ptr(const GFXfont*, uint8_t)':

C:\Users\Rainer\Documents\Arduino\libraries\OneBitDisplay\src\OneBitDisplay.cpp:173:25: error: 'pgm_read_pointer' was not declared in this scope

   return &(((GFXglyph *)pgm_read_pointer(&gfxFont->glyph))[c]);

                         ^~~~~~~~~~~~~~~~

C:\Users\Rainer\Documents\Arduino\libraries\OneBitDisplay\src\OneBitDisplay.cpp:173:25: note: suggested alternative: 'pgm_read_ptr'

   return &(((GFXglyph *)pgm_read_pointer(&gfxFont->glyph))[c]);

                         ^~~~~~~~~~~~~~~~

                         pgm_read_ptr

Using library OneBitDisplay at version 2.0.0 in folder: C:\Users\Rainer\Documents\Arduino\libraries\OneBitDisplay 

Using library BitBang_I2C at version 2.1.5 in folder: C:\Users\Rainer\Documents\Arduino\libraries\BitBang_I2C 

Using library SPI at version 1.0 in folder: C:\Users\Rainer\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.5\libraries\SPI 

Using library Wire at version 1.0 in folder: C:\Users\Rainer\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.5\libraries\Wire 

exit status 1

Error compiling for board Arduino Nano.

80stech, 2022-05-08:
This machine was Windows 10 but I now tried it on a machine with fresh Linux, Arduino IDE and OneBitDisplay and bit bang installed again with the Library manager and similar problem. It must be something simple that I'm doing wrong.

bitbank2, 2022-05-09:
I just pushed a fix; please give it a try.

bitbank2, 2022-05-10:
I didn't do a release, so the zip file and Arduino will still have the old code. Do a git clone or just change the one erroneous line from pgm_read_pointer to pgm_read_ptr

80stech, 2022-05-10:
OK, I changed that line and it compiles and it works now :)
Thanks!

closed 2022-05-10 by 80stech