bitbank2/OneBitDisplay #45: ST7302 Dump speed improvement

kayae, 2023-08-03:
Platform Nano33 BLE 

in OBD.inl
s_tatic int ST7302DumpBuffer(OBDISP *pOBD, uint8_t *pBuffer)_
changing
_RawWrite(pOBD, ucPixels, 3 + (int)(d - ucPixels));_ 
to 
_digitalWrite(pOBD->iDCPin, HIGH); // data mode
digitalWrite(pOBD->iCSPin, LOW);
mySPI->transfer((char*)ucPixels, 3 + (int)(d - ucPixels));
digitalWrite(pOBD->iCSPin, HIGH);_

improves transmission time, from 122ms to 41ms.

Pastant, 2024-02-08:
Could you post your code for the initialization of the ST7302? thx

kayae, 2024-02-09:
***Pins***
#define RST_Disp (6u)
#define DC_Disp (15u)
#define PIN_SPI_MISO (8u)//not used
#define PIN_SPI_MOSI (19u)
#define PIN_SPI_SCK (18u)
#define CS_Disp (14u)

***obd init***
obdSPIInit(&obd, LCD_ST7302, DC_Disp, CS_Disp, RST_Disp, PIN_SPI_MOSI, PIN_SPI_SCK, -1, 0, 0, 0, 8000000);

***obd st7302 setting***
#ifndef WIMPY_MCU
const uint8_t st7302_wenting[] PROGMEM PROGMEM = {
    0x02, 0xD1, 0x01, // Auto Power Control
    //0x02, 0xC0, 0xba , // Gate Voltage setting
    //0x02, 0xCB, 0x00, // VCOMH Setting
    //0x02, 0xB3, 0x94, // VCOMEQ enable
    //0x07, 0xC1, 0x28, 0x28, 0x28, 0x28, 0x14, 0x00, // source voltage VSH
    //0x05, 0xC2, 0x00, 0x00, 0x00, 0x00, // VSL Setting
    //0x0B, 0xB4, 0xE5, 0x66, 0x85, 0xFF, 0xFF, 0x52, 0x85, 0xFF, 0xFF, 0x52, // Gate EQ
    0x03, 0xC7, 0xA6, 0xE9, // OSC Setting
    0x02, 0xB0, 0x64, // Duty Setting
    0x03, 0xB2, 0x01, 0x05, // Frame Rate Setting (Highest for both HPM/LPM)
    0x02, 0x36, 0x00, // Memory Access Mode
    0x02, 0x3A, 0x11, // Data Format
    0x02, 0xB8, 0x09, // Panel Setting
    //0x02, 0xD6, 0x02,
    0x01, 0x11, // Sleep out
    0x02, 0xff, 100, // delay 100
    //0x02, 0xD0, 0x1F, // enable auto power down
    0x01, 0x39, // Low power mode
    0x01, 0x29, // display on
    0x00
};