bitbank2/OneBitDisplay #53: Error "section of 'st7302_wenting' conflicts with previous declaration"

rin67630, 2024-09-24:
Hi,
I tried to compile your unchanged demo for a WemosD1 board on Arduino IDE 2.
I got the error:
`In file included from d:\Activities\3_Maker\libraries\OneBitDisplay\src\OneBitDisplay.cpp:58:
d:\Activities\3_Maker\libraries\OneBitDisplay\src\obd.inl:1064:15: error: section of 'st7302_wenting' conflicts with previous declaration
 1064 | const uint8_t st7302_wenting[] PROGMEM PROGMEM = {
      |               ^~~~~~~~~~~~~~
exit status 1
Compilation error: exit status 1`

Unfortunately the compiler does not mention the other conflicting declaration.
I found only one single occurance of "st7302_wenting" in my development drive, so I really cannot figure where the conflict comes from.

C-Dunham, 2024-10-05:
const uint8_t st7302_wenting[] PROGMEM PROGMEM =

C-Dunham, 2024-10-05:
"const uint8_t st7302_wenting[] PROGMEM PROGMEM = " 
should be 
"const uint8_t st7302_wenting[] PROGMEM = " 

(The PROGMEM keyword should not appear twice)